khoih-prog/MySQL_MariaDB_Generic #2: WiFi Hangs when attemptin to start WiFi (Adafruit M4 Express Airlift Lite board)

gmstuff, 2020-11-05:
Can anybody give me a pointer  please :-)

Using out of the box example (say the Connect_WiFiNINA example in the MySQL_MariaDB_Generic folder)  the code stops dead whenever WiFI.xxxxxxx is attempted

So in the case of this example, the code compiles, prints the first output line to the Serial Monitor then hangs for ever at the line 
  if (WiFi.status() == WL_NO_MODULE)

In the monitor window I get   :
Starting Connect on SAMD51 ADAFRUIT_METRO_M4_AIRLIFT_LITE

I installed the MySQL_MariaDB_Generic library from the Arduino IDE Library manager
I upgraded the firmware on the Adafruit M4 Express Airlift Lite board to be latest 1.7.1 firmware
I have deleted the Arduino IDE
Deleted the sketches\library\MySQL_MariaDB_Generic folder
Reinstalled the 1.8.13 version of the IDE
Reinstalled the Adafruit drivers and boards
And every time I try to compile and run one of the MySQL examples it hangs on any attempt to run the WiFi code.
Using the same  board, when I flip over to one of the standard WiFiNINA example sketches they are able to run and connect to my WiFi network so I know the board and WiFi network is good.

Arduino IDE: 18.13
Board: Adafruit M4 Express Airlift Lite
Sketch ibraries\MySQL_MariaDB_Generic\examples\WiFiNINA\Connect_WiFiNINA\ Connect_WiFiNINA.ino (only changed the SSID and Wifi Password
WiFiNINA Firmware : 1.7.1

Serial monitor output : 

21:56:40.669 -> 
21:56:40.669 -> Starting Connect on SAMD51 ADAFRUIT_METRO_M4_AIRLIFT_LITE

Code snip with the extra Serial.println line is below:
----------------------------------------------------------------------------------------
void setup()
{
  Serial.begin(115200);
  while (!Serial); // wait for serial port to connect

  Serial.println("\nStarting Connect on " + String(BOARD_NAME));

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE)
  {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

Serial.println("Passed WiFi.status....");
------------------------------------------------------------------------------------------

Can anybody  give me any pointers at all?

Thanks so much in advance!

khoih-prog, 2020-11-05:
Hi @gmstuff 

Thanks for using the library.

The [**MySQL_MariaDB_Generic Library**](https://github.com/khoih-prog/MySQL_MariaDB_Generic) is using [**WiFiNINA_Generic Library**](https://github.com/khoih-prog/WiFiNINA_Generic) for boards with WiFiNINA.

It seems that your board has some issue with [WiFiNINA_Generic Library ](https://github.com/khoih-prog/WiFiNINA_Generic) library as you described.

To isolate the issue, please test using the examples in 

1. [**WiFiNINA_Generic examples**](https://github.com/khoih-prog/WiFiNINA_Generic/tree/master/examples) 
2. [**Original WiFiNINA**](https://github.com/arduino-libraries/WiFiNINA)

to see if both are OK with your **ADAFRUIT_METRO_M4_AIRLIFT_LITE** board. Then post the results back here.

I'm sorry I don't have an **ADAFRUIT_METRO_M4_AIRLIFT_LITE** board to test and verify.

The issue might be you haven't connected / modified the file `WiFiNINA_Pinout_Generic.h` to take care of your **ADAFRUIT_METRO_M4_AIRLIFT_LITE** board as warning in compiling process

```
In file included from /home/kh/Arduino/libraries/WiFiNINA_Generic-1.7.1/src/utility/spi_drv.cpp:51:
/home/kh/Arduino/libraries/WiFiNINA_Generic-1.7.1/src/WiFiNINA_Pinout_Generic.h:225:8: warning: #warning You have to modify pin usage according to actual connection for SAMD [-Wcpp]
  225 |       #warning You have to modify pin usage according to actual connection for SAMD
      |        ^~~~~~~
....
```

If you can run OK with  [**Original WiFiNINA**](https://github.com/arduino-libraries/WiFiNINA), I'll modify the `WiFiNINA_Pinout_Generic.h`  to fix the issue.

gmstuff, 2020-11-05:
Thanks for the speedy response !

I saw in the libraries\MySQL_MariaDB_Generic\README.md that the SPI CS pin on the Adafruit SAMD51 boards is pin 36. My problem is I don’t know where to define that :-(  

However you asked I test the WiFiNINA_Generi and the standard WiFiNINA examples: Here are the results

When I use he example : 
libraries\WiFiNINA\examples\SimpleWebServerWiFi\ SimpleWebServerWiFi.ino
And change only the SSD and Password for the wifi
And add in a Serial.print debug line after the WiFi.Status code block 
The code compiles and uploads
And connects to the WiFi network  (Output below)

22:52:14.153 -> Passed the WiFi.status code block
22:52:14.153 -> Attempting to connect to Network named: ASUS
22:52:29.907 -> SSID: ASUS
22:52:29.907 -> IP Address: 192.168.0.36
22:52:29.907 -> signal strength (RSSI):-58 dBm
22:52:29.907 -> To see this page in action, open a browser to http://192.168.0.36

However, When I use he example : 
Sketches\libraries\WiFiNINA_Generic\examples\ SimpleWebServerWiFi \SimpleWebServerWiFi.ino
And change only the SSD and Password for the wifi
And add in a Serial.println debug line after the WiFi.Status code block (code snip below) 
The code compiles and uploads
But hangs  at the same point .Below is the Serial Monitor output 

22:49:18.042 -> 
22:49:18.042 -> Start SimpleWebServerWiFi on SAMD51 ADAFRUIT_METRO_M4_AIRLIFT_LITE

void setup()
{
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial);

  Serial.println("\nStart SimpleWebServerWiFi on " + String(BOARD_NAME));

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE)
  {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

Serial.print("Passed the WiFi.status code block");

  String fv = WiFi.firmwareVersion();

I don’t get any warnings in the compile window:
Sketch uses 35224 bytes (6%) of program storage space. Maximum is 507904 bytes.
Device       : ATSAMD51x19
Version      : v1.1 [Arduino:XYZ] Oct 27 2020 20:26:12
Address      : 0x0
Pages        : 1024
Page Size    : 512 bytes
Total Size   : 512KB
Planes       : 1
Lock Regions : 32
Locked       : none
Security     : false
BOD          : false
BOR          : true
Write 36944 bytes to flash (73 pages)
[==============================] 100% (73/73 pages)
Done in 0.425 seconds
Verify 36944 bytes of flash
[==============================] 100% (73/73 pages)
Verify successful
Done in 0.172 seconds

khoih-prog, 2020-11-05:
That's great.

Can you just add the line

```
#elif defined(ADAFRUIT_METRO_M4_AIRLIFT_LITE)
```

to [**WiFiNINA_Pinout_Generic Line 223**](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/WiFiNINA_Pinout_Generic.h#L223)

It will look like:

```
    #elif defined(SEEED_GROVE_UI_WIRELESS)
    
      #warning You have to modify pin usage according to actual connection for SAMD SEEED_GROVE_UI_WIRELESS
      // To define pin out for WiFiNINA here

      //#define PINS_COUNT           (60u)
      //NINA
      #define NINA_GPIO0  ( 0u)                             //0
      #define NINA_RESETN ( 1u)
      #define NINA_ACK    ( 2u)

      #define SPIWIFI_SS        3   //PIN_SPI1_SS            //3
      #define SPIWIFI_ACK       2   //NINA_ACK               //2
      #define SPIWIFI_RESET     1   //NINA_RESETN            //1      
    
    #elif defined(ADAFRUIT_METRO_M4_AIRLIFT_LITE)    // <======= This is the added line
    
    #else
      #warning You have to modify pin usage according to actual connection for SAMD
      // To define pin out for WiFiNINA here

      //#define PINS_COUNT           (60u)
      //NINA
      #define NINA_GPIO0  (26u)                             //26
      #define NINA_RESETN (27u)
      #define NINA_ACK    (28u)

      #define SPIWIFI_SS       24   //PIN_SPI1_SS            //24
      #define SPIWIFI_ACK      28   //NINA_ACK               //28 
      #define SPIWIFI_RESET    27   //NINA_RESETN            //27

    #endif
  #endif     
```

I think the issue will be fixed.

gmstuff, 2020-11-05:
Ok I am officially impressed!  :-)  :-)   - I beat my head on the wall for two days trying everything I could and got no where! Then I posted tonight an issue and its fixed!

That was a crazy fast response and fix!!

Having made the mod you told me to make, then 
Using the libraries\WiFiNINA_Generic\examples\SimpleWebServerWiFi\SimpleWebServerWiFi.ino
The example connected to my test Wifi network.

This is my output : 
23:21:48.841 -> 
23:21:48.841 -> Start SimpleWebServerWiFi on SAMD51 ADAFRUIT_METRO_M4_AIRLIFT_LITE
23:21:49.596 -> Passed WiFi.status code block
23:21:49.596 -> Attempting to connect to Network named: ASUS
23:21:57.862 -> SSID: ASUS
23:21:57.862 -> IP Address: 192.168.0.36
23:21:57.862 -> signal strength (RSSI):-65 dBm
23:21:57.862 -> To see this page in action, open a browser to http://192.168.0.36

You are *THE*  Man!!!

Thank you !

khoih-prog, 2020-11-05:
Good to know.

I'll update the [**WiFiNINA_Generic Library**](https://github.com/khoih-prog/WiFiNINA_Generic) and certainly have a note on your contribution to add 

1. **SAMD51 ADAFRUIT_METRO_M4_AIRLIFT_LITE**
2. **SAMD51 ADAFRUIT_PYBADGE_AIRLIFT_M4**

to [**WiFiNINA_Generic Library**](https://github.com/khoih-prog/WiFiNINA_Generic) supported list.

Enjoy your project.

Best Regards,

closed 2020-11-05 by khoih-prog

khoih-prog, 2020-11-05:
Dear Gerard,

Your contribution (reporting and collaborating) leads to new [**WiFiNINA_Generic Library v1.7.2**](https://github.com/khoih-prog/WiFiNINA_Generic/releases/tag/v1.7.2) and is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/WiFiNINA_Generic#contributions-and-thanks).

Please don't hesitate to report new issues and/or request for enhancement.

Best Regards,

---

### New in v1.7.2

1. Add support to Adafruit Airlift M4 boards: METRO_M4_AIRLIFT_LITE, PYBADGE_AIRLIFT_M4. Thanks to [Gerard Moorcroft](https://github.com/gmstuff) to report issue [**WiFi Hangs when attempting to start WiFi (Adafruit M4 Express Airlift Lite board)**](https://github.com/khoih-prog/MySQL_MariaDB_Generic/issues/2) leading to this new version.
2. Add WIFININA_GENERIC_VERSION

gmstuff, 2020-11-06:
Hey Man, I am totally not worthy of a mention

I appreciate your help in helping me! Alls I did was ask for help, you guys are the brains and the solution finders!

The SQL_Connector is helping my project hugely, so please, no thanks to me, all thanks to you guys!!