khoih-prog/MySQL_MariaDB_Generic #3: Succesfully query send but ERROR: Timeout waiting for client.

Vasekdvor, 2020-12-10:
Hi i am trying this library and it works it sends query to DB but after every send it print an error:

```
[SQL] MySQL_Packet::read_packet:  ERROR: Timeout waiting for client.
Insert error
```
But the query is succesfull, i think that the library wait for server to send some respond packet on the query, but the server
doesn't sent anything and then it will write this error. Can you help me where should be problem ?

I am using an example basic_insert_esp.ino (just modified the credentials and query)
Server Version = 5.5.57-MariaDB 

I enabled LOGLEVEL 5 and this is what it writes into the the serial:

`================================================
Connecting...
[SQL] Connecting to Server: 192.168.1.100 , Port =  3306
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::read_packet: step 1
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  86 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  86
[SQL] MySQL_Packet::read_packet: step 2
[SQL] MySQL_Packet::read_packet: packet_len=  82
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] MySQL_Packet::read_packet: step 1
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  11 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  11
[SQL] MySQL_Packet::read_packet: step 2
[SQL] MySQL_Packet::read_packet: packet_len=  7
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  0
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
[SQL] Connected. Server Version = 5.5.57-MariaDB
INSERT INTO `mereni_vody`.`spotreba_vody`(`Celkove_mnozstvi_studena_voda`,`Celkove_mnozstvi_tepla_voda`,`Prutok_studena_voda`,`Prutok_tepla_voda`,`Teplota_studene_vody`,`Teplota_teple_vody`,`Mnozstvi_zemniho_plynu`,`Selhani_spojeni_wifi`) VALUES (10.0,10.0,10,10,10,10,10,5)
[SQL] MySQL_Query::execute: Reuse allocated buffer, conn->largest_buffer_size =  279  >  279
[SQL] MySQL_Query::execute: query =  INSERT INTO `mereni_vody`.`spotreba_vody`(`Celkove_mnozstvi_studena_voda`,`Celkove_mnozstvi_tepla_voda`,`Prutok_studena_voda`,`Prutok_tepla_voda`,`Teplota_studene_vody`,`Teplota_teple_vody`,`Mnozstvi_zemniho_plynu`,`Selhani_spojeni_wifi`) VALUES (10.0,10.0,10,10,10,10,10,5)
[SQL] MySQL_Query::execute_query: query =  INSERT INTO `mereni_vody`.`spotreba_vody`(`Celkove_mnozstvi_studena_voda`,`Celkove_mnozstvi_tepla_voda`,`Prutok_studena_voda`,`Prutok_tepla_voda`,`Teplota_studene_vody`,`Teplota_teple_vody`,`Mnozstvi_zemniho_plynu`,`Selhani_spojeni_wifi`) VALUES (10.0,10.0,10,10,10,10,10,5)
[SQL] MySQL_Packet::read_packet: step 1
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: client->stop
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  0
[SQL] MySQL_Packet::read_packet:  ERROR: Timeout waiting for client.
Insert error
[SQL] Disconnected

Sleeping...
================================================
` 
Thanks for your help.

khoih-prog, 2020-12-10:
If **the query is successful** as you're experiencing, there is nothing to worry.

There are too many error messages, causing by the **slow Internet/network/server-response**, just for developers' debugging purposes, and you can ignore them by using LOGLEVEL 0.

This library is intended for those developers, to help us to communicate and locate the bugs, if exist. 

You just disable the debug (LOGLEVEL = 0) when passing to the users.

closed 2020-12-10 by khoih-prog

khoih-prog, 2020-12-10:
Sorry I misunderstood your not-very-clear question

The terminal output must be something like the following

```
Starting Basic_Insert_ESP on ESP32_DEV
Connecting to HueNet1
..Connected to network. My IP address is: 192.168.2.101
Connecting to SQL Server @ 192.168.2.112, Port = 5698
User = invited-guest, PW = the-invited-guest, DB = test_arduino
Connecting...
[SQL] Connecting to Server: 192.168.2.112 , Port =  5698
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  103
[SQL] MySQL_Packet::read_packet: packet_len=  99
[SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  103
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  11
[SQL] MySQL_Packet::read_packet: packet_len=  7
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  0
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
[SQL] Connected. Server Version = 5.5.5-10.3.25-MariaDB-0+deb10u1
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Query::execute: Reuse allocated buffer, conn->largest_buffer_size =  103  >  80
[SQL] MySQL_Query::execute: query =  INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Query::execute_query: query =  INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  13
[SQL] MySQL_Packet::read_packet: packet_len=  9
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  0
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
Data Inserted.
[SQL] Disconnected

Sleeping...
```
---
There are many things can be wrong, mostly in rthe way you create the MariaDB database.

Only you can check, such as 

1. database user/password must be correct and user has enough privileges to read/write to database

```
char user[]         = "invited-guest";              // MySQL user login username
char password[]     = "the-invited-guest";          // MySQL user login password
```
2. Did you create the database correctly (name, table, etc.)

```
char default_database[] = "test_arduino";           //"test_arduino";
char default_table[]    = "hello_arduino";          //"test_arduino";
```

3. Start with the easiest examples to duplicate the working results before modifying them for your purposes.

---

I don't think I can help any further because these are so basic database steps you must master before you can to anything else with it.

Good Luck,

Vasekdvor, 2020-12-10:
I created db and table based on your recomendations, using an SQL query by Dr. Charles Bell and use the really basic input on esp8266, with your recommended settings and its still writes the ERROR: Timeout waiting for client.

I even tried to change: 
`#define MYSQL_DATA_TIMEOUT  6000`
to 10000 milis. without any change, still getting this error.

Privileges are:

 ```
SELECT
 INSERT
 UPDATE
 DELETE
 FILE
 CREATE
 ALTER
 INDEX
 DROP
 CREATE TEMPORARY TABLES
 SHOW VIEW
 CREATE ROUTINE
 ALTER ROUTINE
 EXECUTE
 CREATE VIEW
 EVENT
 TRIGGER
```

khoih-prog, 2020-12-10:
There are many basic things you can check

1. [**A Note About Testing Access to MySQL**](https://github.com/ChuckBell/MySQL_Connector_Arduino/wiki/Troubleshooting#a-note-about-testing-access-to-mysql)

2. Change `bind-address` in config

```
# Instead of skip-networking the default is now to listen only on
# localhost which is more compatible and is not less secure.
#bind-address            = 127.0.0.1
bind-address            = 192.168.xxx.yyy
```
3. [**Basic Insert**](https://github.com/ChuckBell/MySQL_Connector_Arduino/wiki/Examples#basic-insert-basic_insertino)

4. [**How to Install MariaDB**](https://raspberrytips.com/install-mariadb-raspberry-pi/)

5. Test to connect from other machine using ssh, then access MariaDB using exactly the same INSERT command to see if OK

6. Check why you have the extra \` \` around the database/table/column, etc. **Possible some wrong settings in your text editor or Windows system.**

It must be something like 
**INSERT INTO mereni_vody.spotreba_vody**
not 
**INSERT INTO \`mereni_vody\`.\`spotreba_vody\`**

Correct:

> INSERT INTO mereni_vody.spotreba_vody (Celkove_mnozstvi_studena_voda,Celkove_mnozstvi_tepla_voda,Prutok_studena_voda,Prutok_tepla_voda,Teplota_studene_vody,Teplota_teple_vody,Mnozstvi_zemniho_plynu,Selhani_spojeni_wifi) VALUES (10.0,10.0,10,10,10,10,10,5)

not 

> INSERT INTO \`mereni_vody\`.\`spotreba_vody\`(\`Celkove_mnozstvi_studena_voda\`,\`Celkove_mnozstvi_tepla_voda\`,\`Prutok_studena_voda\`,\`Prutok_tepla_voda\`,\`Teplota_studene_vody\`,\`Teplota_teple_vody\`,\`Mnozstvi_zemniho_plynu\`,\`Selhani_spojeni_wifi\`) VALUES (10.0,10.0,10,10,10,10,10,5)

Vasekdvor, 2020-12-11:
I will try to answer to all of your recommendation points:

1. Its not my case, my SQL server (MariaDB) is running on NAS and i am trying to connect to it directly from ESP8266

2. I could try that, but i think that there will be no difference

3. I already tried that, and its still the same

4. My MariaDB server was preinstalled on NAS and working fine with any other software

5. I tried that from my PC using an SQLyog and from the web using PHPMyAdmin - it works fine

6. These extra ` are inserted intentionally because without them i experiencing some problems on many Mysql servers, even on
hosted ones from third party, these ` are used by PHPMyAdmin in any generated SQL command and every time i am using it it works fine and better then without them. 

These recomendations you made are not related to question, in the question i wrote that inserting sql command are made succefully and the new row are inserted into the DB and with correct values, so the SQL query statement must be OK and have a right syntax, because if not, it will simply won't insert anything into DB.

The problem is the connection timeout which is too small or library doesn't even receive any OK_packet from DB, and that cause the problems with timeout.

There is something wrong with DB responding,but i don't know what and what should i try to make it better.

I tried to change engine from MyISAM to InnoDB and now its much better, 95% of sended queries are ending with:

`[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
Data Inserted.`

Which is nice and we should say that its the problem, but i think that i only make it faster to executing sql queries in DB and then we fit into the max timeout. So there is definitely a short timeout and if i change the timeout values in .h files there is no difference, which should be. Or wait at least some time and then expecting some response from DB.

khoih-prog, 2020-12-11:
If your first post is as clear as it is now, there must be no misunderstood because of `your not-very-clear question`.
I have no psychic power to guess what's wrong with your installation, and just can read between the lines that

```
Insert error
```
and think that there was no success in INSERT command.

As I said

> If the query is successful as you're experiencing, there is nothing to worry.

Anyway, at least you found out the issue is mainly your database excessive responding time.