khoih-prog/MySQL_MariaDB_Generic #6: Select query double

Reletiv123, 2021-05-25:
So i have been trying to get a double value back from a select query but i always seem to get a integer. I even looked in the library files if i can change it myself. But i can not get it to work.

khoih-prog, 2021-05-25:
Thanks for using the library.

Please follow the instructions in [**Issue: Bug report**](https://github.com/khoih-prog/MySQL_MariaDB_Generic/issues/new?assignees=&labels=&template=bug_report.md&title=) and post all the information, especially the [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example).

Otherwise, you issue will be ignored and closed.

Reletiv123, 2021-05-25:
`#include <MySQL_Generic_WiFi.h>

// Network Info
char ssid[] = "xx";
char pass[] = "xx";

// Database Adress
IPAddress server_addr(xx, xx, xx, xx);
uint16_t server_port = 3306;

// Database Account
char user[]         = "xx";
char password[]     = "xx";

//Database Name
char default_database[] = "xx_weerstation";
char default_table[]    = "weerstation";

//Default values
float temperatuur = 0;
float luchtvochtigheid = 0;

//Query
String query1 = String("SELECT temperatuur FROM xx_weerstation.weerstation ORDER BY id DESC LIMIT 1;");
String query2 = String("SELECT luchtvochtigheid FROM xx_weerstation.weerstation ORDER BY id DESC LIMIT 1;");

// Create an instance of the cursor passing in the connection
MySQL_Connection conn((Client *)&client);
MySQL_Query sql_query = MySQL_Query(&conn);

void setup()
{
  // Open Serial
  Serial.begin(115200);  

  // Begin WiFi section
  Serial.println(String("Connecting to ") + ssid);
  WiFi.begin(ssid, pass);
  
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.print(".");
  }

  // print out info about the connection:
  Serial.print("Connected to network. My IP address is: ");
  Serial.println(WiFi.localIP());

  // test mysql connection
  Serial.print("Connecting to SQL Server @ ");
  Serial.print(server_addr);
  Serial.println(String(", Port = ") + server_port);
  Serial.println(String("User = ") + user + String(", PW = ") + password + String(", DB = ") + default_database);
}

void runQuery(void)
{
  row_values *row = NULL;
  float temp = 0;

  Serial.println("1) Demonstrating using a dynamically allocated query.");
  // Initiate the query class instance
  MySQL_Query query_mem = MySQL_Query(&conn);
  
  // Execute the query
  Serial.println(query1);

  // Execute the query
  // KH, check if valid before fetching
  if ( !query_mem.execute(query1.c_str()) )
  {
    Serial.println("Querying error");
    return;
  }
  
  // Fetch the columns (required) but we don't use them.
  column_names *columns = query_mem.get_columns();

  // Read the row (we are only expecting the one)
  do 
  {
    row = query_mem.get_next_row();
    
    if (row != NULL) 
    {
      temp = atol(row->values[0]);
    }
  } while (row != NULL);

  // Show the result
  Serial.print("  Temperatuur = ");
  Serial.println(temp);
  temperatuur = temp;
  delay(500);

  // Now we close the cursor to free any memory
  sql_query.close();

}

void runQuery2(void)
{
  row_values *row = NULL;
  float humi = 0;

  Serial.println("1) Demonstrating using a dynamically allocated query.");
  // Initiate the query class instance
  MySQL_Query query_mem = MySQL_Query(&conn);
  
  // Execute the query
  Serial.println(query2);

  // Execute the query
  // KH, check if valid before fetching
  if ( !query_mem.execute(query2.c_str()) )
  {
    Serial.println("Querying error");
    return;
  }
  
  // Fetch the columns (required) but we don't use them.
  column_names *columns = query_mem.get_columns();

  // Read the row (we are only expecting the one)
  do 
  {
     row = query_mem.get_next_row();
    
    if (row != NULL) 
    {
      humi = atol(row->values[0]);
    }
  } while (row != NULL);

  // Show the result
  Serial.print("  Temperatuur = ");
  Serial.println(humi);
  luchtvochtigheid = humi;
  delay(500);

  // Now we close the cursor to free any memory
  sql_query.close();

}

void loop()
{
  Serial.println("Connecting...");
  if (conn.connectNonBlocking(server_addr, server_port, user, password) != RESULT_FAIL)
  {
    runQuery();
    runQuery2();
    conn.close();                     // close the connection
    
    Serial.print("Temp: " + String(temperatuur) + "Celcius");
    Serial.print("Humi: " + String(luchtvochtigheid) + " %");
  } 
  else 
  {
    Serial.println("\nConnect failed. Trying again on next iteration.");
  }

  
  Serial.println("\nSleeping...");
  Serial.println("================================================");
  delay(100000);
}`

here is my code. this is the result i get:
[image]
but this is the result i expect, the way its in my database, with the decimals:
[image]

the issue is replicable by uploading the code to a esp8266. u can get my database info if u send me a message. dont want to put my login cridentials on the open here.

Reletiv123, 2021-05-26:
i see that the code chat function didn't work as intended. excuse me. if u need anything else im here

khoih-prog, 2021-06-02:
Hi @Reletiv123 

Thanks for using the library.

The issue you got happens because you didn't change the example to adapt to your use case.

For example

```
temp = atol(row->values[0]);

...

humi = atol(row->values[0]);
```

Please check [**atol = Convert string to long integer**](https://www.cplusplus.com/reference/cstdlib/atol/)
You have to use [**atof = Convert string to double**](https://www.cplusplus.com/reference/cstdlib/atof/) for your purpose

```
temp = atof(row->values[0]);

...

humi = atof(row->values[0]);
```

Good Luck,

closed 2021-06-02 by khoih-prog