khoih-prog/MySQL_MariaDB_Generic #8: NativeEthernet support for Teensy 4.1

graham999au, 2021-08-30:
### Describe the bug

I am getting compiler errors when using the library with NativeEthernet.h and NativeEthernetUdp.h.
NativeEthernet is the library for the add-on ethernet connection for Teensy 4.1 boards.

### Steps to Reproduce

#include <NativeEthernet.h>
#include <NativeEthernetUdp.h>
#include <MySQL_Generic_Ethernet.h>

### Expected behavior

I would like to use this library to connect to MySQL.

### Actual behavior

Compiler errors

### Debug and AT-command log (if applicable)

### Screenshots

If applicable, add screenshots to help explain your problem.

### Information

Please ensure to specify the following:

* Arduino IDE version (e.g. 1.8.13) or Platform.io version
Teensyduino
Teensy 4.1 wth SPI ethernet
  
### Example

Context:

### Additional context

Arduino: 1.8.15 (Mac OS X), TD: 1.54, Board: "Teensy 4.1, Serial, 600 MHz, Faster, US English"

In file included from /Users/grahamalston/Documents/Arduino/Flex-Mediator-Teensy-V0.5/Flex-Mediator-Teensy-V0.5.ino:19:0:
/Users/grahamalston/Documents/Arduino/libraries/MySQL_MariaDB_Generic/src/MySQL_Generic_Ethernet.h:81:4: warning: #warning Using Ethernet lib [-Wcpp]
   #warning Using Ethernet lib
    ^
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Dns.cpp.o: In function `DNSClient::begin(IPAddress const&)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Dns.cpp:48: multiple definition of `DNSClient::begin(IPAddress const&)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeDns.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeDns.cpp:10: first defined here
/Applications/Teensyduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Dns.cpp.o: In function `DNSClient::begin(IPAddress const&)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Dns.cpp:48: multiple definition of `DNSClient::inet_aton(char const*, IPAddress&)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeDns.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeDns.cpp:28: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Dns.cpp.o: In function `DNSClient::begin(IPAddress const&)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Dns.cpp:48: multiple definition of `DNSClient::getHostByName(char const*, IPAddress&, unsigned short)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeDns.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeDns.cpp:28: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Ethernet.cpp.o: In function `EthernetClass::begin(unsigned char*, unsigned long, unsigned long)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Ethernet.cpp:30: multiple definition of `EthernetClass::begin(unsigned char*, unsigned long, unsigned long)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeEthernet.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeEthernet.cpp:64: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Ethernet.cpp.o: In function `Print::flush()':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Ethernet.cpp:81: multiple definition of `EthernetClass::begin(unsigned char*, IPAddress, IPAddress, IPAddress)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeEthernet.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeEthernet.cpp:389: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Ethernet.cpp.o: In function `Print::flush()':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Ethernet.cpp:81: multiple definition of `EthernetClass::begin(unsigned char*, IPAddress, IPAddress)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeEthernet.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeEthernet.cpp:389: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Ethernet.cpp.o: In function `Print::flush()':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Ethernet.cpp:81: multiple definition of `EthernetClass::begin(unsigned char*, IPAddress)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeEthernet.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeEthernet.cpp:389: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/Ethernet.cpp.o: In function `Print::flush()':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/Ethernet.cpp:81: multiple definition of `EthernetClass::begin(unsigned char*, IPAddress, IPAddress, IPAddress, IPAddress)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/NativeEthernet.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/NativeEthernet.cpp:389: first defined here
[... 198 lines cut ...]
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/socket.cpp.o: In function `EthernetClass::socketBegin(unsigned char, unsigned short)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/socket.cpp:399: multiple definition of `EthernetClass::socketRecvAvailable(unsigned char)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/socket.cpp.o: In function `EthernetClass::socketBegin(unsigned char, unsigned short)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/socket.cpp:399: multiple definition of `EthernetClass::socketPeek(unsigned char)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/socket.cpp.o: In function `EthernetClass::socketBegin(unsigned char, unsigned short)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/socket.cpp:399: multiple definition of `EthernetClass::socketSend(unsigned char, unsigned char const*, unsigned short)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/Ethernet/socket.cpp.o: In function `EthernetClass::socketBegin(unsigned char, unsigned short)':
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/socket.cpp:399: multiple definition of `EthernetClass::socketSendAvailable(unsigned char)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
[... cut ...]
/Users/grahamalston/Documents/Arduino/libraries/Ethernet/src/socket.cpp:399: multiple definition of `EthernetClass::socketStartUDP(unsigned char, unsigned char*, unsigned short)'
/var/folders/fc/5yx1qmm15ql65bp9b6qh2nxh0000gn/T/arduino_build_933505/libraries/NativeEthernet/Nativesocket.cpp.o:/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/NativeEthernet/src/Nativesocket.cpp:44: first defined here
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 4.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

khoih-prog, 2021-08-30:
As you know, the `Teensy 4.1` with `NativeEthernet` is not currently supported by this library.

I'll try to add the support in the future, certainly if time permitted.

Thanks for your interests in the library,

Regards,

khoih-prog, 2021-08-30:
I'm currently working on this enhancement and it's working now.

Will have a new release within tomorrow to support Teensy4.1 and NativeEthernet

```
Starting Basic_Insert on Teensy 4.1 , with NativeEthernet
MySQL_MariaDB_Generic v1.2.0
[SQL] =========================================
[SQL] Default SPI pinout:
[SQL] MOSI: 11
[SQL] MISO: 12
[SQL] SCK: 13
[SQL] SS: 10
[SQL] =========================================
[SQL] Board : Teensy 4.1 , setCsPin: 10
[SQL] =========================================
[SQL] Currently Used SPI pinout:
[SQL] MOSI: 11
[SQL] MISO: 12
[SQL] SCK: 13
[SQL] SS: 10
[SQL] =========================================
Using mac index = 10
Connected! IP address: 192.168.2.87
Connecting to SQL Server @ 192.168.2.112 , Port = 5698
User = invited-guest , PW = the-invited-guest , DB = test_arduino
Connecting...
[SQL] Connecting to Server: 192.168.2.112 , Port =  5698
[SQL] Connect OK. Try reading packets
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 72
[SQL] Connected. Server Version = 5.5.5-10.3.29-MariaDB-0+deb10u1
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
Data Inserted.
[SQL] Disconnected

Sleeping...
================================================
Connecting...
[SQL] Connecting to Server: 192.168.2.112 , Port =  5698
[SQL] Connect OK. Try reading packets
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 72
[SQL] Connected. Server Version = 5.5.5-10.3.29-MariaDB-0+deb10u1
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
Data Inserted.
[SQL] Disconnected

Sleeping...
================================================
```

Cheers,

graham999au, 2021-08-31:
That is fantastic...THANK YOU! I will test and report any issues.

khoih-prog, 2021-08-31:
Hi @graham999au 

The new release [**MySQL_MariaDB_Generic** v1.3.0](https://github.com/khoih-prog/MySQL_MariaDB_Generic/releases/tag/v1.3.0) has just been published.

Your contribution, by requesting enhancement to support Teensy 4.1 NativeEthernet, has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/MySQL_MariaDB_Generic#contributions-and-thanks)

Also have a look at my preliminary test result [**Complex_Select on Teensy 4.1 with NativeEthernet**](https://github.com/khoih-prog/MySQL_MariaDB_Generic#16-complex_select-on-teensy-41-with-nativeethernet)

Please have more thorough tests and report any issue here.

Best Regards,

---

### Release v1.3.0

1. Add support to **Teensy 4.1 using NativeEthernet**

closed 2021-08-31 by khoih-prog

khoih-prog, 2021-09-05:
Hi @graham999au 

Could you please try again using [**QNEthernet** library](https://github.com/ssilverman/QNEthernet) ?

In case you haven't tried it, have a quick look at this very good [**New lwIP-based Ethernet library for Teensy 4.1**](https://forum.pjrc.com/threads/68066-New-lwIP-based-Ethernet-library-for-Teensy-4-1) and [**New-and-Modified-Libraries-to-support-Teensy-4-1-QNEthernet**](https://forum.pjrc.com/threads/68117-New-and-Modified-Libraries-to-support-Teensy-4-1-QNEthernet)

---

### Major Release v1.4.0

1. Add support to [QNEthernet Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Fix bug in non-blocking connection
3. Add new feature to permit using either server's hostname, such as `your_account.duckdns.org`, or server's IPAddress, such as IPAddress(192,168,2,112)
4. Update examples with new features, bug-fixes, etc.

graham999au, 2021-09-06:
Hi,

I am not set up to test IwIP sorry, but I have installed QNEthernet and it is working. Where do you define UDP_TX_PACKET_MAX_SIZE?

khoih-prog, 2021-09-06:
> I am not set up to test IwIP sorry, but I have installed QNEthernet and it is working. 

Good. You don't need to install **lwip** separately as it's already included in **QNEthernet**

> Where do you define UDP_TX_PACKET_MAX_SIZE?

Where and why you need that? 

It's normally defined in **Ethernet/WiFi** library, such as

```
#define UDP_TX_PACKET_MAX_SIZE 24
```
I find it nowhere in **QNEthernet**. You can post an issue there and ask for support if necessary.