khoih-prog/MySQL_MariaDB_Generic #11: Please describe a way, how to install the library for ESP8266

hasenradball, 2021-11-22:
Wenn using this Library and install it like the others in PIO I got this horrible error mess:

Whats going on here?

I want to use the MariaDB Connector for the ESP8266....

```
rocessing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 72 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MySQL_MariaDB_Generic> 1.5.1
|   |-- <WiFiNINA_Generic> 1.8.13
|   |   |-- <SPI> 1.0
|   |-- <ESP8266_AT_WebServer> 1.4.0
|   |   |-- <Functional-Vlpp> 1.0.2
|   |-- <STM32duino LwIP> 2.1.2
|   |-- <STM32Ethernet> 1.2.0
|   |   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |   |-- <STM32duino LwIP> 2.1.2
|   |-- <EthernetENC> 2.0.1
|   |   |-- <SPI> 1.0
|   |-- <WiFiEspAT> 1.3.1
|   |-- <Ethernet2> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <Ethernet3> 1.5.5
|   |   |-- <SPI> 1.0
|   |-- <Ethernet> 2.0.0
|   |   |-- <SPI> 1.0
|   |-- <EthernetLarge> 2.0.0
|   |   |-- <SPI> 1.0
|   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |-- <STM32duino LwIP> 2.1.2
|   |-- <SPI> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |   |-- <STM32duino LwIP> 2.1.2
|   |   |-- <STM32duino LwIP> 2.1.2
|-- <movingAVG> 0.0.1
|-- <cVCC> 1.1.1
|-- <cWIFI> 2.3.6
|   |-- <ESP8266WiFi> 1.0
|   |   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |   |-- <STM32duino LwIP> 2.1.2
|   |   |-- <STM32duino LwIP> 2.1.2
|   |-- <WiFiEspAT> 1.3.1
|   |-- <DBG_Print> 2.0.4
|   |   |-- <ESP8266WiFi> 1.0
|   |   |   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |   |   |-- <STM32duino LwIP> 2.1.2
|   |   |   |-- <STM32duino LwIP> 2.1.2
|   |   |-- <WiFiEspAT> 1.3.1
|-- <DBG_Print> 2.0.4
|   |-- <ESP8266WiFi> 1.0
|   |   |-- <QNEthernet> 0.11.0-snapshot+sha.abab89f
|   |   |   |-- <STM32duino LwIP> 2.1.2
|   |   |-- <STM32duino LwIP> 2.1.2
|   |-- <WiFiEspAT> 1.3.1
|-- <ESP8266WiFi> 1.0
[... 132 lines cut ...]
*
*******************************************************************

   42 | #include "stm32_def.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp12e\lib39c\STM32Ethernet\EthernetClient.cpp.o] Error 1
In file included from .pio\libdeps\esp12e\STM32Ethernet\src\EthernetClient.h:7,
                 from .pio\libdeps\esp12e\STM32Ethernet\src\STM32Ethernet.h:6,
                 from .pio\libdeps\esp12e\STM32Ethernet\src\EthernetServer.cpp:5:
.pio\libdeps\esp12e\STM32Ethernet\src\utility/stm32_eth.h:42:10: fatal error: stm32_def.h: No such file or directory

*******************************************************************
* Looking for stm32_def.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stm32_def.h"
* Web  > https://platformio.org/lib/search?query=header:stm32_def.h
*
*******************************************************************

   42 | #include "stm32_def.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp12e\lib39c\STM32Ethernet\EthernetServer.cpp.o] Error 1
```

khoih-prog, 2021-11-22:
Hi,

Thanks for your interest in the library.

As you know, using PIO requires a little more experiences and knowledge than Arduino IDE to understand and manually modify the settings. Moreover, the current PIO version is **not smart enough to select the correct library, according the board,** necessary to use with this library, which is written mainly for **multi-platforms** and Arduino IDE.

**Just try using Arduino IDE and understand why it can selects correctly the libraries to use, no matter boards / platforms you're using.**

This library's `platformio.ini` is also written to **include all possible dependent libraries for all the boards with the expectation that PIO users will know and modify to ignore the incorrect ones**. It's better than include nothing and users has to do research to know which libraries are needed and include them.

Just understand which libraries to be used for your board, you'll know that you can **ignore some unused libraries**, such as adding to your `platformio.ini`

```
lib_ignore = 
   STM32AsyncTCP
   STM32duino LwIP
   STM32Ethernet
   WebServer_WT32_ETH01
```

Good Luck,

closed 2021-11-22 by khoih-prog

hasenradball, 2021-11-25:
Hi thanks,

I was not aware how to handle this library :-)
Thank you for your help.

:-)