khoih-prog/MySQL_MariaDB_Generic #13: Basic insert ESP8266 MySQL_Packet::wait_for_bytes: client->stop

DavidAntonin, 2021-12-29:
I filled up user names /passwords and compiled simple insert on ESP8266.
Here is a log from serial:
Connecting...
[SQL] Connecting to Server: 192.168.2.100 , Port =  3306
[SQL] connected = 1
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  110
[SQL] MySQL_Packet::read_packet: packet_len=  106
[SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  110
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 66
[SQL] MySQL_Packet::wait_for_bytes: client->stop
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  0
[SQL] MySQL_Packet::read_packet:  ERROR: Timeout waiting for client.
[SQL] Can't connect. Error reading auth packets

Connect failed. Trying again on next iteration.

Y0Sam, 2021-12-30:
I have the same issue with an ESP07. On a ESP12 all works well using the exact same code,
Using the Arduino IDE, with different memory settings (via the tools menu) and other possible influencing settings.

The serial output:
Connecting...
[SQL] Connecting to Server: 192.168.13.199 , Port =  3306
:close
:ur 1
:dsrcv 0
:del
:ref 1
[SQL] connected = 1
[SQL] Connect OK. Try reading packets
:rn 111
:rcl pb=0x3fff11bc sz=111
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  111
[SQL] MySQL_Packet::read_packet: packet_len=  107
:c0 1, 111
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 67
:wr 67 0
[SQL] MySQL_Packet::wait_for_bytes: client->stop
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  0
[SQL] MySQL_Packet::read_packet:  ERROR: Timeout waiting for client.
[SQL] Can't connect. Error reading auth packets

Connect failed. Trying again on next iteration.

Build output:
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 243504          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 29301   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1716  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3148  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26576 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 277669 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31440 bytes (38%) of dynamic memory, leaving 50480 bytes for local variables. Maximum is 81920 bytes.
esptool.py v3.0
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:29:06:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 281824 bytes to 205658...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 281824 bytes (205658 compressed) at 0x00000000 in 5.1 seconds (effective 440.0 kbit/s)...
Hash of data verified.
(For ESP12 Auto-detected Flash size: 4MB)

khoih-prog, 2021-12-30:
Hi @Y0Sam 

> I have the same issue with an ESP07. On a ESP12 all works well using the exact same code,

I don't have the ESP07 to duplicate the issue. So you're on your own.

khoih-prog, 2021-12-30:
Hi @DavidAntonin 

Please provide the [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) and all the related info as requested in [**Issue: Bug report**](https://github.com/khoih-prog/MySQL_MariaDB_Generic/issues/new?assignees=&labels=&template=bug_report.md&title=)

For example

* Arduino IDE version (e.g. 1.8.19)
* `ESP8266`  Core Version (e.g. ESP8266 core v3.0.2)
* Board: ESP8266_NODEMCU_ESP12E
* etc.

Without problem duplication, there is no way anyone can solve the vague issue.

DavidAntonin, 2021-12-31:
Arduino IDE version (e.g. 1.8.19)
ESP8266 Core Version (e.g. ESP8266 core v3.0.2)
Board: NodeMCU 1.0 (ESP12E module)
MySQL_MariaDB_Generic v1.5.2

code:
`#if ! (ESP8266 || ESP32 )
  #error This code is intended to run on the ESP8266/ESP32 platform! Please check your Tools->Board setting
#endif

#include "Credentials.h"

#define MYSQL_DEBUG_PORT      Serial

// Debug Level from 0 to 4
#define _MYSQL_LOGLEVEL_      4

#include <MySQL_Generic.h>

#define USING_HOST_NAME     false

#if USING_HOST_NAME
  // Optional using hostname, and Ethernet built-in DNS lookup
  char server[] = "your_account.ddns.net"; // change to your server's hostname/URL
#else
  IPAddress server(192, 168, 2, 100);
#endif

uint16_t server_port = 3306;    //3306;

char default_database[] = "test";           //"test_arduino";
char default_table[]    = "tabulka";          //"test_arduino";

String default_value    = "Hello, Arduino!"; 

// Sample query
//String INSERT_SQL = String("INSERT INTO ") + default_database + "." + default_table + " (text) VALUES ('" + default_value + "')";
String INSERT_SQL = "INSERT INTO test.tabulka (text) VALUES ('test z test1')";
MySQL_Connection conn((Client *)&client);

MySQL_Query *query_mem;

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  MYSQL_DISPLAY1("\nStarting Basic_Insert_ESP on", ARDUINO_BOARD);
  MYSQL_DISPLAY(MYSQL_MARIADB_GENERIC_VERSION);

  // Begin WiFi section
  MYSQL_DISPLAY1("Connecting to", ssid);
  
  WiFi.begin(ssid, pass);
  
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    MYSQL_DISPLAY0(".");
  }

  // print out info about the connection:
  MYSQL_DISPLAY1("Connected to network. My IP address is:", WiFi.localIP());

  MYSQL_DISPLAY3("Connecting to SQL Server @", server, ", Port =", server_port);
  MYSQL_DISPLAY5("User =", user, ", PW =", password, ", DB =", default_database);
}

void runInsert()
{
  // Initiate the query class instance
  MySQL_Query query_mem = MySQL_Query(&conn);

  if (conn.connected())
  {
    MYSQL_DISPLAY(INSERT_SQL);
    
    // Execute the query
    // KH, check if valid before fetching
    if ( !query_mem.execute(INSERT_SQL.c_str()) )
    {
      MYSQL_DISPLAY("Insert error");
    }
    else
    {
      MYSQL_DISPLAY("Data Inserted.");
    }
  }
  else
  {
    MYSQL_DISPLAY("Disconnected from Server. Can't insert.");
  }
}

void loop()
{
  MYSQL_DISPLAY("Connecting...");
  
  //if (conn.connect(server, server_port, user, password))
  if (conn.connectNonBlocking(server, server_port, user, password) != RESULT_FAIL)
  {
    delay(500);
    runInsert();
    conn.close();                     // close the connection
  } 
  else 
  {
    MYSQL_DISPLAY("\nConnect failed. Trying again on next iteration.");
  }

  MYSQL_DISPLAY("\nSleeping...");
  MYSQL_DISPLAY("================================================");
 
  delay(60000);
}`

List from console:
`Starting Basic_Insert_ESP on ESP8266_NODEMCU_ESP12E
MySQL_MariaDB_Generic v1.5.2
Connecting to NasNet
.........Connected to network. My IP address is: 192.168.2.32
Connecting to SQL Server @ 192.168.2.100 , Port = 3306
User = arduino , PW = Hesl0-databaze , DB = test
Connecting...
[SQL] Connecting to Server: 192.168.2.100 , Port =  3306
[SQL] connected = 1
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  110
[SQL] MySQL_Packet::read_packet: packet_len=  106
[SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  110
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 66
[SQL] MySQL_Packet::wait_for_bytes: client->stop
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  0
[SQL] MySQL_Packet::read_packet:  ERROR: Timeout waiting for client.
[SQL] Can't connect. Error reading auth packets

Connect failed. Trying again on next iteration.

Sleeping...
`

I tried this insert with basic php script and it is working:

```
<?php
        $server="192.168.2.100";
		$user="arduino";
		$pass="Hesl0-databaze";
	   	
		$connection = mysql_connect($server, $user, $pass);
        	if (!$connection) {
	    	die('MySQL ERROR: ' . mysql_error());
		}
        
        $query ="INSERT INTO test.tabulka (text) VALUES ('test z test1')";
        mysql_query($query,$connection);
        mysql_close($connection);

?>
```

khoih-prog, 2021-12-31:
Hi @DavidAntonin 

I tested using [Basic_Insert_ESP](https://github.com/khoih-prog/MySQL_MariaDB_Generic/tree/master/examples/WiFi/Basic_Insert_ESP) with **ESP8266_NODEMCU_ESP12E** without any issue.

I think there must be some issue with your database or your modified code

---

Debug Terminal output => **OK**

```
Starting Basic_Insert_ESP on ESP8266_NODEMCU_ESP12E
MySQL_MariaDB_Generic v1.5.2
Connecting to HueNet1
...........Connected to network. My IP address is: 192.168.2.93
Connecting to SQL Server @ 192.168.2.112 , Port = 5698
User = invited-guest , PW = the-invited-guest , DB = test_arduino
Connecting...
[SQL] Connecting to Server: 192.168.2.112 , Port =  5698
[SQL] connected = 1
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  103
[SQL] MySQL_Packet::read_packet: packet_len=  99
[SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  103
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 72
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  11
[SQL] MySQL_Packet::read_packet: packet_len=  7
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  0
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
[SQL] Connected. Server Version = 5.5.5-10.3.31-MariaDB-0+deb10u1
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Query::execute: Reuse allocated buffer, conn->largest_buffer_size =  103  >  80
[SQL] MySQL_Query::execute: query =  INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Query::execute_query: query =  INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  13
[SQL] MySQL_Packet::read_packet: packet_len=  9
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  0
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
Data Inserted.
[SQL] Disconnected

Sleeping...
================================================

```

closed 2021-12-31 by khoih-prog