khoih-prog/MySQL_MariaDB_Generic #22: suggest replacing all files xxxx_Impl.h by xxxx.cpp

pabloandresm, 2022-03-12:
May I suggest suggest replacing all files xxxx_Impl.h by xxxx.cpp?

Having all code in .h header files creates linking errors if you "#include <MySQL_Generic>" from different .cpp modules of the same project, because the linker will find several times functions with the same name.

Having a proper .h corresponding to a proper .cpp, you can include the .h several times from everywhere without causing lining problems.

It happened to me.

Just a suggestion.

khoih-prog, 2022-03-12:
Have a look at

1. [**HOWTO Fix Multiple Definitions Linker Error**](https://github.com/khoih-prog/RP2040_Slow_PWM#howto-fix-multiple-definitions-linker-error) or similar issues.
2. [Different behaviour using the src_cpp or src_h lib #80](https://github.com/khoih-prog/ESPAsync_WiFiManager/discussions/80)

I forget to convert this to `h-only` library with example similar to [**multiFileProject**](https://github.com/khoih-prog/RP2040_Slow_PWM/tree/main/examples/multiFileProject) how to deal with the issue.

Will publish new release to deal with this, but never going back to `.cpp-style`, unless absolutely necessary.

pabloandresm, 2022-03-13:
thank you for that info.
I ll read those links.
thanks

khoih-prog, 2022-03-13:
Hi @pabloandresm 

The new [**MySQL_MariaDB_Generic releases v1.7.0**](https://github.com/khoih-prog/MySQL_MariaDB_Generic/releases/tag/v1.7.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/MySQL_MariaDB_Generic#contributions-and-thanks)

Please test one of the new examples, such as [**multiFileProject_Ethernet**](https://github.com/khoih-prog/MySQL_MariaDB_Generic/tree/master/examples/Ethernet/multiFileProject_Ethernet), to see how to do for multiple-file projects to avoid `Multiple Definitions` Linker Error

Best Regards,

PS: 

I'm excited and so busy to follow up with your suggestions, but still wish you provide **many more ;-))**

---

### Releases v1.7.0

1. Convert to `h-only` style
2. Add `multiFileProject` examples to demo for multiple-file projects

closed 2022-03-13 by khoih-prog

pabloandresm, 2022-03-13:
thank you very much.

I will try to keep up with the suggestions and fixes.

Are you going to publish it soon? so I can install it via Arduino.

Let me try the 1.7.0 and stress it and I will let you know.

thank you very much

khoih-prog, 2022-03-13:
Already published. You can install from Arduino or PlatformIO now.

pabloandresm, 2022-03-13:
strange.....arduino shows me the latest one as 1.6.1, and not the 1.7.0

khoih-prog, 2022-03-13:
It takes some time to propagate to Library Manager (some hours ??). You can download directly from the GitHub, then unzip.

PS:

You have a good talent to finding bugs, especially using your stress tests. I wish you can use that to spots the bugs in my other libraries.

pabloandresm, 2022-03-13:
yes, sure. Which other libraries you need to test?

one detail I forgot to mention.....since 1.6.0, when I update this library on Arduino IDE, it asks me if I want to update just this library or the dependencies also (it shows me a list of 5 dependency libraries). If I choose all dependencies, it successfully updates ONLY this library and fails with an error. Maybe this is caused by some dependency you marked that are ment to be for platformio?

I will screenshot it next time.

khoih-prog, 2022-03-13:
Because this library is written to support too many boards, shields, libraries, architectures and platforms, I have to add all of them to the list. Many users complaint long ago if not. So better extra than nothing.

Sometimes, the Arduino IDE and/or PIO is not intelligent enough to select the correct libraries for that architectures / platform.

So, you have to manually install the necessary libraries, at least you know the list.

khoih-prog, 2022-03-13:
You can see my full list of 130+ libraries in [KH Libraries](https://registry.platformio.org/search?q=author%3A%22Khoi+Hoang%22&p=1)

Just jump into any library you're interested and possibly will use in your project, then click `Homepage` to jump to GitHub to download / test. All are in Arduino IDE Library Manager as well and you can use the name to install.

khoih-prog, 2022-03-13:
@pabloandresm 

I'm going to invite you to a Private Repo to deal with the libraries' testing as this is not the suitable place and topics.

khoih-prog, 2022-03-13:
Already invited and waiting for your acceptance.