khoih-prog/MySQL_MariaDB_Generic #24: [Question]: Call Stored procedure possible

hasenradball, 2022-06-17:
Hi,

is it actually possible to call a stored Procedureby this library?

best regards
Frank

khoih-prog, 2022-06-17:
Hi @hasenradball 

Yes, if you can manually use the stored Procedure (your new command) without error and working normally in your DB, you can use it in the code.

If you're doing OK, please share your experience here.

Best Regards,

closed 2022-06-17 by khoih-prog

hasenradball, 2022-06-17:
Hi,

what is the function for this?
I did‘ t saw it.

khoih-prog, 2022-06-17:
I totally misunderstood your question. It's better you'd write in German, then we can use Google Translate.

You also have to post some more detailed of your use-case or your code so that we can have better picture.

Normally, any function, supported by MySQL or MariaDB, can be used in your code.

For example, to use **INSERT** command

https://github.com/khoih-prog/MySQL_MariaDB_Generic/blob/e479474f013b1162c1fd7283928d099ec5030617/examples/Ethernet/Complex_Insert/Complex_Insert.ino#L93

hasenradball, 2022-06-17:
Hallo, 

kein Problem, ich schreibe in Deutsch.
Ich wollte folgendes fragen.
Kann die Library eine in der Datenbank vorhandene „ stored procedure“ ausführen?
Damit meine ich keine SQL Statements wie select oder insert, sondern den Aufruf eine in der Datenbank gespeicherte Procedure.

Es ist effizienter eine Insert routine als stored procedure in der Datenbank anzulegen und diese von außen als Funktion mit Parametern aufzurufen.

khoih-prog, 2022-06-17:
Using Google Translate. Please check and modify if not accurate

> Hello,
> 
> no problem, I write in German.
> I wanted to ask the following.
> Can the library execute a "stored procedure" in the database?
> By that I don't mean SQL statements like select or insert, but calling a procedure stored in the database.
> 
> It is more efficient to create an insert routine as a stored procedure in the database and call it from the outside as a function with parameters.

khoih-prog, 2022-06-17:
The best thing is to try to manually implement it first, with the DB. After testing and OK, I think you can duplicate the `stored procedure` in your code.

This library is dumb, and just passes any `command` you have in your code to the DB. Nothing more.

hasenradball, 2022-06-17:
Hi again,

I have a stored procedure in the DB, but I want just call it , not duplicate it in the code!
😉

hasenradball, 2022-06-18:
Hi,

I am sorry it should work as you set just call it like:
```
// call Procedure: => CALL `your_procedure`(MRT, zeit, MGR1, MGR2, MGR3, MGR4);
snprintf(SqlStr, SQL_LEN, "CALL `procedure`('%d', '%s', '%5.2f', '%5.2f', '%5.2f', '%5.2f')", value1, value2, value3, value4, value5, value6;
```
I will try it 👍

hasenradball, 2022-06-20:
Hi,
works perfect see the result:
[image]

khoih-prog, 2022-06-20:
Hi @hasenradball 

**Marvellous**. If possible, could you please write an example, as an PR, to contribute as a contributor and to benefit other users.

If not familiar with creating PR, just post here and I'll update the library.

Best Regards,

hasenradball, 2022-06-21:
Hi @khoih-prog,

I will test it today and can then mak a short example.
I am not sure If you had adapt it, because I had a local version of your lib.
I stripped it down only to work with ESP8266 and ESP32.

We can discuss if I had the example :-)

hasenradball, 2022-06-23:
Hi,
no problem I will help you, but Iam a little bit running out of time before my holiday.

The change is very simple.

1) write a stored procedure in you DB
2) test the stored procedure in the DB
3) the just adapt the typical SQL_INSERT String --> to "Call `your_strored_procedure` ....

here you will find the stored procedure "addState" in MarianDB:
[image]

The table "Switch_State" looks like:
[image]

The just adapt the String INSERT_SQL to:
```
// Query itself
  // Sample query
  // e.g.: => Call addState(time, state);
  String INSERT_SQL = 
    String("Call `addState`(" + String(time) + ", " + state + ");" );
  DBG__PRINT(INSERT_SQL);
```

My Function looks like:

```
void MariaDB_addState(time_t time, bool state) {
  // Wifi Client for Connection to IP
  WiFiClient client;

  // create a connector
  MySQL_Connection conn(&client);

  // create a Query class instance
  MySQL_Query query_mem = MySQL_Query(&conn);

  // Query itself
  // Sample query
  // e.g.: => Call addState(time, state);
  String INSERT_SQL = 
    String("Call `addState`(" + String(time) + ", " + state + ");" );
  DBG__PRINT(INSERT_SQL);

  Wifi wlan{SSID, PASSWORD};
  if (wlan.Wifi_Connect()) {
    DBG__PRINT("WLAN established!");
    if (conn.connect(DB_HOST, DB_PORT, DB_USER, DB_PASSWORD, DB_DATABASE)) {
      DBG__PRINT("DB Connected!");
      if (!query_mem.execute(INSERT_SQL.c_str())) {
        DBG__PRINT("ERROR: INSERT failed!");
      }
      else {
        DBG__PRINT("INFO: INSERT successful!");
      }
      DBG__PRINT("INFO: close connector...");
    }
    else {
      DBG__PRINT("ERROR: DB Connection failed!");
    }
  }
  conn.close();
  client.stop();
}
```

That`s it.
Can you deal with it?
If not we can disscuss in two weeks. :-)

PS: you see it works fine

[image]

khoih-prog, 2022-07-05:
Hi @hasenradball 

I think this issue has **enough details** for anyone would like to start using `Stored Procedures`.
Therefore, I won't spend time to write an example and instructions.

Thanks and Regards,