khoih-prog/MySQL_MariaDB_Generic #26: Can't connect to XAMPP's MariaDb

Elfagor, 2022-09-28:
### Describe the bug

A clear and concise description of what the bug is.

### Steps to Reproduce

Steps to reproduce the behavior. Including the [MRE](https://stackoverflow.com/help/minimal-reproducible-example) sketches

### Expected behavior

A clear and concise description of what you expected to happen.

### Actual behavior

A clear and concise description of what you expected to happen.

### Debug and AT-command log (if applicable)

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Information

Please ensure to specify the following:

* Arduino IDE version (e.g. 1.8.13) or Platform.io version
* `ESP8266`,`ESP32` or `STM32` Core Version (e.g. ESP8266 core v2.7.4, ESP32 v1.0.5 or STM32 v1.9.0)
* Contextual information (e.g. what you were trying to achieve)
* Simplest possible steps to reproduce
* Anything that might be relevant in your opinion, such as:
  * Operating system (Windows, Ubuntu, etc.) and the output of `uname -a`
  * Network configuration
  
### Example

```
Arduino IDE version: 1.8.13
ESP32 Core Version 1.0.5
OS: Ubuntu 20.04 LTS
Linux xy-Inspiron-3593 5.4.0-66-generic #74-Ubuntu SMP Wed Jan 27 22:54:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Context:
I encountered an endless loop while trying to connect to Local WiFi.

Steps to reproduce:
1. ...
2. ...
3. ...
4. ...
```

### Additional context

Add any other context about the problem here.

Elfagor, 2022-09-28:
I was trying your library in order to connect Arduino to a MariaDb (within XAMPP). 
Connection to an external mysql server works like a charm.
Connection to XAMPP's mariadb server also work correctly from command line mysql and from php, either with localhost and with a specific ip address.

calling:
MySQL_Connection *conn;
[...]
conn = new MySQL_Connection(client);
[...]
conn->connect(*data->networkData->server_addr, data->networkData->port, username, password)) 

with the right information provided always gives a bad handshake error.

[SQL] Connecting to Server: 192.168.1.147 , Port =  3306
[SQL] connected = 1
[SQL] Connect OK. Try reading packets
[SQL] MySQL_Packet::read_packet: step 1
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  93 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  93
[SQL] MySQL_Packet::read_packet: step 2
[SQL] MySQL_Packet::read_packet: packet_len=  89
[SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  93
[SQL] MySQL_Packet::read_packet: exit
[SQL] Try parsing packets
5.5.5-10.4.20-MariaDB
[SQL] Try send_authentication packets
[SQL] Writing this_buffer, size_send = 42
[SQL] MySQL_Packet::read_packet: step 1
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  0 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: Num bytes=  26 , need bytes=  4
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  26
[SQL] MySQL_Packet::read_packet: step 2
[SQL] MySQL_Packet::read_packet: packet_len=  22
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_packet_type: packet type=  255
[SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_ERROR_PACKET
[SQL] Error:  22  =
Bad handshake.
Connection failed.

Any suggestion to resolve the problem? Thanks in advance.

Elfagor

khoih-prog, 2022-09-28:
Hi @Elfagor 

Thanks for your interest and usage of the library.

I've just installed and latest `XAMPP` on Ubuntu 22.04 LTS and test OK after

1. Create user "invited-guest" with PW "the-invited-guest" according to example

```
char user[]     = "invited-guest";              // MySQL user login username
char password[] = "the-invited-guest";          // MySQL user login password
```

2. Download and Import `world` database from [**Other MySQL Documentation**](https://dev.mysql.com/doc/index-other.html) to MariaDB of XAMPP

3. Run the example [Basic_Select](https://github.com/khoih-prog/MySQL_MariaDB_Generic/tree/master/examples/Ethernet/Basic_Select) and OK
The `Server Version = 5.5.5-10.4.24-MariaDB`

#### Debug level 1

```
Starting Basic_Select on RASPBERRY_PI_PICO , with W5x00 using Ethernet_Generic Library on SPI0/SPI
MySQL_MariaDB_Generic v1.7.2
[SQL] =========================================
[SQL] Default SPI pinout:
[SQL] MOSI: 19
[SQL] MISO: 16
[SQL] SCK: 18
[SQL] SS: 17
[SQL] =========================================
[SQL] RPIPICO setCsPin: 17
[SQL] Currently Used SPI pinout:
[SQL] MOSI: 19
[SQL] MISO: 16
[SQL] SCK: 18
[SQL] SS: 17
[SQL] =========================
Using mac index = 17
Connected! IP address: 192.168.2.98
Connecting to SQL Server @ 192.168.2.40 , Port = 3306
User = invited-guest , PW = the-invited-guest , DB = world
Connecting...
1) Demonstrating using a dynamically allocated query.
SELECT population FROM world.city WHERE name = 'Toronto'
  Toronto pop = 688275
2) Demonstrating using a local, global query.
SELECT population FROM world.city WHERE name = 'Toronto'
  Toronto pop = 688275
  Toronto pop increased by 11725 = 700000
[SQL] Disconnected

Sleeping...
```

#### Full debug level 4

```
Starting Basic_Select on RASPBERRY_PI_PICO , with W5x00 using Ethernet_Generic Library on SPI0/SPI
MySQL_MariaDB_Generic v1.7.2
[SQL] =========================================
[SQL] Default SPI pinout:
[SQL] MOSI: 19
[SQL] MISO: 16
[SQL] SCK: 18
[SQL] SS: 17
[SQL] =========================================
[SQL] RPIPICO setCsPin: 17
[SQL] Currently Used SPI pinout:
[SQL] MOSI: 19
[SQL] MISO: 16
[... 129 lines cut ...]
[SQL] MySQL_Packet::read_packet: exit
[SQL] MySQL_Packet::get_lcb_len: read_len=  1
[SQL] MySQL_Query::read_string: offset =  4
[SQL] MySQL_Query::read_string: len =  6 len_bytes = 1
[SQL] MySQL_Query::read_string: len =  6 conn->buffer size = 93
[SQL] MySQL_Query::read_string: str =  688275
[SQL] MySQL_Query::get_next_row: get_row_values
[SQL] MySQL_Query::get_row_values: get_row
[SQL] MySQL_Query::get_row: read_packet
[SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  9
[SQL] MySQL_Packet::read_packet: packet_len=  5
[SQL] MySQL_Packet::read_packet: exit
  Toronto pop = 688275
  Toronto pop increased by 11725 = 700000
[SQL] Disconnected

Sleeping...
================================================
```

---

I think you have some issue with **installation, settings, etc**. As I don't have tine to help you here, please post the issue on appropriate Forum, such as Arduino, etc.to ask for help from many experts there.

Good Luck,

closed 2022-09-28 by khoih-prog

khoih-prog, 2022-09-28:
Another user had the same issue long time ago with `XAMPP`, and already fixed it.

Have a look at https://github.com/khoih-prog/MySQL_MariaDB_Generic/issues/1#issuecomment-683234605

> It's working now ! Only, I don't know what was wrong earlier. I reinstalled xampp and created a new user and granted all privileges and it works like before !
> 
> Strange ! Don't know what mistake I had made earlier

Elfagor, 2022-09-28:
Thanks for your super-fast response. Just to know: what privileges did you granted to your users? and what type of external connection did you enabled to your xampp sql server? for example? did you set mysql_native_password, mariadb_native_password or whatever as "plugin" in the view user (SELECT * FROM `user`)? I'm starting to think the problem is the type of plugin adopted.

In short, could you do "SELECT * FROM ..." both global_priv and user tables/views in your mysql DB? I hope just copying your configurations would be sufficient to have everything working fine! :-D

Thanks in advance!!!

khoih-prog, 2022-09-28:
Sorry I don't have time to explain these mundane details as I just did it naturally and even don't remember anything now. XAMPP or not is just basic and normal MariaDB. By using it, you must possess certain DB knowledge, to install and configure.

You have to do your basic research, then modify the configs, and try until OK. 

Ask on some Forum if problem persists.

Elfagor, 2022-09-28:
ok, thanks anyhow.

Elfagor, 2022-09-28:
Ok, it worked like a charm. Honestly, I do not know what the problem had been with auth details... anyhow. It works, that's great! Thank you very much!

khoih-prog, 2022-09-28:
Great to know.

It'll be very helpful if you can **note and describe the steps** you fixed the issue, so that other users can benefit.

The library can be better thanks to the contribution of users like you.

We, library writers with tunnel-vision, seldom have and understand the similar problems the users meet.