khoih-prog/MySQL_MariaDB_Generic #27: Reboot ESP8266

MaurizioBarbera, 2022-11-12:
### Describe the bug

Reboot NodeMCU ESP3266

### Steps to Reproduce

I change error level esp = 0/3 and mysql = 0/3 when try execute query the board loop of reboot 
but set error level 4 on mysql e esp is all okay

### Expected behavior

A clear and concise description of what you expected to happen.

### Actual behavior

A clear and concise description of what you expected to happen.

### Debug and AT-command log (if applicable)

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Information

Please ensure to specify the following:

* Arduino IDE version 2.0.1
* `ESP8266` 3.0.2
* Board config: NodeMCU 1.0 ESP-12E

khoih-prog, 2022-11-12:
Hi @MaurizioBarbera 

Sorry there is something wrong with your board,system,  installations, etc.

#### Debug Terminal

```
// Debug Level from 0 to 4
#define _MYSQL_LOGLEVEL_      1
```

```
Starting Basic_Insert_ESP on ESP8266_NODEMCU_ESP12E
MySQL_MariaDB_Generic v1.7.2
Connecting to HueNet1
...........Connected to network. My IP address is: 192.168.2.126
Connecting to SQL Server @ myacount.ddns.net , Port = 5698
User = invited-guest , PW = the-invited-guest , DB = test_arduino
Connecting...
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
Data Inserted.
[SQL] Disconnected

Sleeping...
================================================
Connecting...
INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')
Data Inserted.
[SQL] Disconnected

Sleeping...
================================================

```

closed 2022-11-12 by khoih-prog

MaurizioBarbera, 2022-11-12:
I saw that you closed the issue, I hope you can reply to this email

Hi, I'm sorry but I was not clear in the question I presented to you.
I tried to perform a select on a table on a local NAS with 147 columns and over 1000 rows.
setting the mysql log level to 4 manages to read all the columns / rows but if I set a lower level it goes into error.

LOG from the serial

#define _ESP_AT_LOGLEVEL_       1
#define _MYSQL_LOGLEVEL_        2

22:15:00.622 -> Starting Query_Results_WiFi on ESP8266_NODEMCU_ESP12E
22:15:00.622 -> MySQL_MariaDB_Generic v1.7.2
22:15:00.622 -> Connecting to FRITZ!Box 7530 MU
22:15:01.116 -> .......Connected to network. My IP address is: 192.168.1.51
22:15:04.876 -> Connecting to SQL Server @ 192.168.1.5 , Port = 3306
22:15:04.876 -> User = admin , PW = xxxxxx
22:15:04.876 -> Connecting...
22:15:04.876 -> [SQL] Connecting to Server: 192.168.1.5 , Port =  3306
22:15:05.488 -> [SQL] Connected. Server Version = 5.5.47-MariaDB
22:15:05.973 ->
22:15:05.973 -> Running SELECT and printing results
22:15:05.973 -> SELECT * FROM Pegaso2013.Zone LIMIT 6
22:15:05.973 -> [SQL] Packet error.
22:15:05.973 -> [SQL] Bad mojo. EOF found reading column header.
22:15:05.973 ->
22:15:05.973 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:15:06.004 ->
22:15:06.004 -> Exception (28):
22:15:06.004 -> epc1=0x402025ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
22:15:06.004 ->
22:15:06.004 -> >>>stack>>>
22:15:06.004 ->
22:15:06.004 -> ctx: cont
22:15:06.004 -> sp: 3ffffcc0 end: 3fffffc0 offset: 0190
22:15:06.004 -> 3ffffe50:  3fffdad0 3ffe8c82 3ffeeb90 402025c4
22:15:06.004 -> 3ffffe60:  ffffff00 00000013 00000013 3ffefc6c
22:15:06.004 -> 3ffffe70:  3fff0554 3fff059c 3fff05dc 3fff061c
22:15:06.036 -> 3ffffe80:  3fff0664 3fff06ac 3fff0764 3fff02a4
22:15:06.036 -> 3ffffe90:  3fff02ec 3ffefe4c 3fff039c 3fff03e4
22:15:06.036 -> 3ffffea0:  3fff0434 3fff04d4 00000000 00000000
22:15:06.036 -> 3ffffeb0:  00000000 00000000 00000000 00000000
22:15:06.036 -> 3ffffec0:  00000000 00000000 00000000 00000000
22:15:06.036 -> 3ffffed0:  00000000 00000000 00000000 00000000
22:15:06.036 -> 3ffffee0:  00000000 00000000 00000000 00000000
22:15:06.036 -> 3ffffef0:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff00:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff10:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff20:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff30:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff40:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff50:  00000000 00000000 00000000 00000000
22:15:06.068 -> 3fffff60:  00000000 00000000 00000000 ffffffff
22:15:06.100 -> 3fffff70:  ffffffff 3ffeea48 000001f4 4020471c
22:15:06.100 -> 3fffff80:  40204dd9 000001f4 3ffeec1c 3ffeec1c
22:15:06.100 -> 3fffff90:  3fffdad0 3ffeea48 3ffeeb90 4020268a
22:15:06.100 -> 3fffffa0:  3fffdad0 00000000 3ffeec08 4020483c
22:15:06.100 -> 3fffffb0:  feefeffe feefeffe 3ffe8610 40100c25
22:15:06.100 -> <<<stack<<<
2:15:06.100 ->
22:15:06.100 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:15:06.138 ->
22:15:06.138 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:15:06.138 ->
22:15:06.138 -> load 0x4010f000, len 3460, room 16
22:15:06.138 -> tail 4
22:15:06.138 -> chksum 0xcc
22:15:06.138 -> load 0x3fff20b8, len 40, room 4
22:15:06.138 -> tail 4
22:15:06.138 -> chksum 0xc9
22:15:06.138 -> csum 0xc9
22:15:06.138 -> v00045e60
22:15:06.138 -> ~ld
22:15:06.241 ->
22:15:06.241 -> Starting Query_Results_WiFi on ESP8266_NODEMCU_ESP12E
22:15:06.241 -> MySQL_MariaDB_Generic v1.7.2
22:15:06.241 -> Connecting to FRITZ!Box 7530 MU
22:15:06.708 -> .......Connected to network. My IP address is: 192.168.1.51
22:15:10.457 -> Connecting to SQL Server @ 192.168.1.5 , Port = 3306
22:15:10.457 -> User = admin , PW = xxxxxx
22:15:10.457 -> Connecting...

khoih-prog, 2022-11-12:
Hi @MaurizioBarbera 

Even now you didn't post your problem code, so that anybody can duplicate the issue to know if this is your coding (client, server, etc.) problem or not. You've lowered your credit score and I won't touch this issue anymore, unless you prove there is a bug in the library.

You're on your own and hope time is also precious to you.

Good Luck,

khoih-prog, 2022-11-13:
FYI, the test result when 

```
// Debug Level from 0 to 4
#define _MYSQL_LOGLEVEL_      1
```

#### Terminal

```
Starting Query_Results_WiFi on ESP8266_NODEMCU_ESP12E
MySQL_MariaDB_Generic v1.7.2
Connecting to HueNet
.......Connected to network. My IP address is: 192.168.2.126
Connecting to SQL Server @ 192.168.2.112 , Port = 5698
User = invited-guest , PW = the-invited-guest
Connecting...

Running SELECT and printing results
SELECT * FROM world.city LIMIT 6
ID, Name, CountryCode, District, Population
1, Kabul, AFG, Kabol, 1780000
2, Qandahar, AFG, Qandahar, 237500
3, Herat, AFG, Herat, 186800
4, Mazar-e-Sharif, AFG, Balkh, 127800
5, Amsterdam, NLD, Noord-Holland, 731200
6, Rotterdam, NLD, Zuid-Holland, 593321
[SQL] Disconnected

Sleeping...
================================================
```

MaurizioBarbera, 2022-11-13:
Hi, time is precious for anyone. Sorry again for my lack of clarity. The code I used is that of the example: Basic_Select_WiFi modifying only the WiFi credentials, IP Mysql, Login MySql, Query Select and LogLevel. After several tests I tried to reduce the number of columns to read and everything seems to work fine. Thank you for the time you have dedicated to me 
PS: if you deem it necessary I can send you privately (creating personalized credentials on my server) to carry out tests