khoih-prog/MySQL_MariaDB_Generic #28: MySQL  Not detecting Wifi - Adafruit M4 Feather with a Featherwing on top

gmstuff, 2022-11-28:
Any help most welcome as I am beating my head on the wall with this! :-(

Thanks so much in advance !!

Short version of the problem: 

- MySQL Exmple (Connect_WiFiNINA) unable to detect WiFi Module., 
- Error in Serial monitor: 
- But the WiFi Module functions correctly with a different sketch
- MySQL Connect_WiFiNINA.ino : No conneciton
- Adafruit ConnectWithWEP.ino : Connects ok.
- 

Thus hardware and network params are ok. 

I am guessing its something to do with the pin configuration 
Adafruit web site states the M4 Feather pins for their Featherwing should be : 
https://learn.adafruit.com/adafruit-airlift-featherwing-esp32-wifi-co-processor-featherwing/arduino
#define SPIWIFI       SPI  // The SPI port
#define SPIWIFI_SS    13   // Chip select pin
#define ESP32_RESETN  12   // Reset pin
#define SPIWIFI_ACK   11   // a.k.a BUSY or READY pin
#define ESP32_GPIO0   -1

The Adafruit (non MySQL) variant has a WiFi.setPins command where the pins are defined.   It would appear the Defines in the MySQL example sketch sets the board type but I cant find the pin config (the file  \src\WiFiNINA_Pinout_Generic.h seems to be the pin layout but I am not sure......

### Steps to Reproduce

Adafruit M4 feather  with an Adafruit Feather wing installed on top
Download standard MySQL Sketch   - Connect_WiFiNINA.ino, in the serial monitor the below is displayed: 
18:28:21.194 -> 
18:28:21.194 -> Starting Connect_WiFiNINA on SAMD51 ADAFRUIT_FEATHER_M4_EXPRESS
18:28:21.194 -> MySQL_MariaDB_Generic v1.7.2
18:28:21.963 -> Communication with WiFi module failed!

### Expected behavior
Without me changing the hardware but simply uploading the default Adafruit ConnectWithWEP example it works with the below in the Serial Monitor
18:32:15.645 -> Startiung....ConnectWithWEP  Raw Adafruit Example with just SSID/Password changes
18:32:16.435 -> Attempting to connect to WEP network, SSID: xxxxxxxxxx
18:32:36.537 -> You're connected to the networkSSID: xxxxxxxxxx
18:32:36.537 -> BSSID: A0:04:60:09:EA:07
18:32:36.537 -> signal strength (RSSI):-43
18:32:36.537 -> Encryption Type:4
18:32:36.537 -> 
18:32:36.537 -> IP Address: 192.168.0.54
18:32:36.537 -> 192.168.0.54
18:32:36.537 -> MAC address: 7C:9E:BD:DC:D2:54

### Information

Arduino IDE : 1.8.16
MySQL Library : 1.7.2
Ruining on Windows 10

I attached a copy of the working sketch : ConnectWithWEP_RawAdafruitExample_V1a.zip
and the non working sketch  : Connect_WiFiNINA.zip

On github, in the section "3. To use WiFiNINA module/shield"
I tried using the params as sugegsted 

defines.h
I left 'as is'  (no change)  it has the two lines at th ebootm : 
#define USING_WIFININA_GENERIC        true
#define USING_WIFININA                false

and in the sketch changed the line 
from : #include <MySQL_Generic.h>
to : #include <MySQL_Generic_WiFiNINA.h>

When I do that the compiler errors on  the line: 'MySQL_Connection conn((Client *)&client);'
with
'MySQL_Connection' does not name a type

```
[ConnectWithWEP_RawAdafruitExample_V1a.zip](https://github.com/khoih-prog/MySQL_MariaDB_Generic/files/10099458/ConnectWithWEP_RawAdafruitExample_V1a.zip)

Context:
These are the distilled files,   I am migrating the project from a Adafruit M4 Airlift to an Adafruit M4 Express Feather  when encountering problems, so I started working backwards until I got to the base examples and couldn't get the M4 Feather to connect to WifI  :-(

khoih-prog, 2022-11-28:
Hi @gmstuff 

It's good that you found your way out of the WiFiNINA mess, which was created by too many different boards from Adafruit, etc., each is using a different connection. That's why the `WiFiNINA_Pinout_Generic.h` was created, but never be up-to-date enough

The less intrusive way you can use is to use the original Adafruit WiFiNINA library, tailored for that M4 board by removing

https://github.com/khoih-prog/MySQL_MariaDB_Generic/blob/9428be5573c9347ae4ede683c002e078356eb981/examples/WiFiNINA/Basic_Insert_WiFiNINA/defines.h#L353

or any

```
#define USING_WIFININA_GENERIC       true
```
from your code.

Check the reason hereafter

https://github.com/khoih-prog/MySQL_MariaDB_Generic/blob/9428be5573c9347ae4ede683c002e078356eb981/src/MySQL_Generic_WiFiNINA.h#L42-L54

It's beneficial that you share the experience here for other users.

Anyway, I'm closing the issue here as it was solved.

Good Luck,

closed 2022-11-28 by khoih-prog

gmstuff, 2022-11-28:
Thanks for the pointers!

And thanks for a great library!  It does all I need of it!

https://github.com/khoih-prog/MySQL_MariaDB_Generic/blob/9428be5573c9347ae4ede683c002e078356eb981/examples/WiFiNINA/Basic_Insert_WiFiNINA/defines.h#L353

https://github.com/khoih-prog/MySQL_MariaDB_Generic/blob/9428be5573c9347ae4ede683c002e078356eb981/src/MySQL_Generic_WiFiNINA.h#L42-L54