khoih-prog/MySQL_MariaDB_Generic #29: MySQL With Feather - Failing after 10 - 20 inserts - always on the line "if (conn.connectNonBlocking(server, server_port, user, password) != RESULT_FAIL)   {"

gmstuff, 2022-12-06:
### Describe the bug

Any help / pointers much appreciated!!  I have spent a few days trying all sorts to identify! (Though I am no expert!!)

After about 10 to 20 records are correctly inserted, the library starts reporting [SQL] connected = 0

When the line if (conn.connectNonBlocking(server, server_port, user, password) != RESULT_FAIL)
executes and control is transfered to that code, the code executes for the first 10 inserts, then between 10 and 20 inserts later the library reports "-> [SQL] connected = 0"

it *SEEMS* like something funky is happening because on one of the MySQL debug lines in the serial output it shows some random characters (overrun?) where it prints the contents of the insert command (in bold below)

19:35:42.062 -> [SQL] MySQL_Query::execute_query: query =  INSERT INTO dev_poolcontrol.pc_log (Proc, Status, Message) VALUES ('proc','Stat','RecCount:16')**Ȋ.⸮N**
Whereas a few lines in the code prior to that I print the same char array to the Serial Monitor and encapsulate with <  and > to make sure no trailing chars and the char array prints without the random characters. see below line
19:35:42.062 -> Query  = <INSERT INTO dev_poolcontrol.pc_log (Proc, Status, Message) VALUES ('proc','Stat','RecCount:16')>

### Steps to Reproduce
I used a Adafruit M4 Feather with a WiFi Feather wing (using the WiFiNINA drivers)  - See my issue posted 11/28 #28 that I thought I had worked around :-(   
I hacked about with the file : WiFiNINA_Pinout_Generic.h  and basically just put in the valiues I *THINK* are needed
I used fundamentally one of the MySQL examples but put a bunch of debug code in to try and narrow down the issue.

After 10 - 20 inserts the library reports SQL Disconnected

### Expected behavior

I have another M4 Device (Adafruit M4 Airlift lite) humming away using the MySQl driver at the same time and it writes a record every 10 seconds (its been going for a few days) so we know the MySQL server (on the same local network ) is running

### Information

Please ensure to specify the following:

Arduino 1.8.16
Running on Windows 10
MySQL Library 1.7.2
I have attached the full sketch and defines.h and the edited pinout.h file
The edited WiFiNINA_Pinout_Generic.h   
The serial monitor output
The key points in that file are at time slot 19:35:41.218 - through 19.35.47.900
Up to **RecCount:16**   are working records, all subsequent from **RecCount:17** fail.

I pasted those lines here in case its easier
19:35:41.218 -> Connecting...
19:35:41.218 -> 147 -In Loop About to insert record**RecCount:16**>
19:35:41.218 -> 151 Is connected
19:35:41.218 -> [SQL] Connecting to Server: 192.168.0.10 , Port =  3306
19:35:41.218 -> [SQL] connected = 1

….multiple  debug output messages (some of which I added)

19:35:41.546 -> --------- 160 in if(conn.connectNonBlocking.....  - about to insert
19:35:42.062 -> In RunInsert
19:35:42.062 -> 204 In RunInsert: Insert data <INSERT INTO %s.%s (Proc, Status, Message) VALUES ('%s',%d,%s)>
… more debug message
19:35:42.343 -> 227 In RunInsert: Inserted
19:35:42.343 -> --------- 163 in if(conn.connectNonBlocking.....  - finsiehd  insert
19:35:42.859 -> 173 -Insert over**RecCount:16**>
19:35:42.859 -> 
19:35:42.859 -> Sleeping...
19:35:42.859 -> ================================================

So that one work all good.

9:35:42.859 -> ================================================
19:35:47.876 -> Connecting...
19:35:47.876 -> 147 -In Loop About to insert record**RecCount:17**>
19:35:47.876 -> 151 Is connected
19:35:47.876 -> [SQL] Connecting to Server: 192.168.0.10 , Port =  3306
19:35:57.861 -> [SQL] connected = 0
19:35:57.861 -> [SQL] Can't connect. Retry # 1
19:36:07.900 -> [SQL] connected = 0
19:36:07.900 -> [SQL] Can't connect. Retry # 2
19:36:17.887 -> [SQL] connected = 0

Files Attached

[Connect_WiFiNINA_Reduced_Defines_V2_failsAfter10Records_v2.zip](https://github.com/khoih-prog/MySQL_MariaDB_Generic/files/10159670/Connect_WiFiNINA_Reduced_Defines_V2_failsAfter10Records_v2.zip)
[WiFiNINA_Pinout_Generic_Edited.zip](https://github.com/khoih-prog/MySQL_MariaDB_Generic/files/10159671/WiFiNINA_Pinout_Generic_Edited.zip)

gmstuff, 2022-12-07:
Update:
Using an 'out of the box' example : Complex_Insert_WiFiNINA
And only make changes to the SSID/Password, The MySQL Server IP address and port num
Setting DEBUG = 4 in the defines.h

The example successfully runs for 10 loops
then fails each time thereafter

Reuplaoding the same code - I get the same results
The MySQL server responds to other inserts from other devices throughout

Output below: **Note the random characters on the [SQL} debug lines from the library ( I bolded to highlight)**
this is the first insert the 11th insert attempt is below that

21:12:26.759 -> Starting Complex_Insert_WiFiNINA on SAMD51 ADAFRUIT_FEATHER_M4_EXPRESS
21:12:26.759 -> MySQL_MariaDB_Generic v1.7.2
21:12:27.556 -> Please upgrade the firmware
21:12:27.556 -> Attempting to connect to SSID: xxxxxxxxxxx
21:12:34.744 -> SSID: xxxxxxxxx IP Address: 192.168.0.54
21:12:34.744 -> Signal strength (RSSI): -41 dBm
21:12:34.744 -> Connecting to SQL Server @ 192.168.0.10 , Port = 3306
21:12:34.744 -> User = admin , PW = admin , DB = dev_poolcontrol
21:12:34.744 -> Connecting...
21:12:34.744 -> [SQL] Connecting to Server: 192.168.0.10 , Port =  3306
21:12:34.791 -> [SQL] connected = 1
21:12:34.791 -> [SQL] Connect OK. Try reading packets
21:12:35.072 -> [SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  78
21:12:35.072 -> [SQL] MySQL_Packet::read_packet: packet_len=  74
21:12:35.072 -> [SQL] MySQL_Packet::read_packet: First time allocate buffer, size =  78
21:12:35.072 -> [SQL] MySQL_Packet::read_packet: exit
21:12:35.072 -> [SQL] Try parsing packets
21:12:35.072 -> [SQL] Try send_authentication packets
21:12:35.072 -> [SQL] Writing this_buffer, size_send = 64
21:12:35.399 -> [SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  11
21:12:35.399 -> [SQL] MySQL_Packet::read_packet: packet_len=  7
21:12:35.399 -> [SQL] MySQL_Packet::read_packet: exit
21:12:35.399 -> [SQL] MySQL_Packet::get_packet_type: packet type=  0
21:12:35.399 -> [SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
21:12:35.399 -> [SQL] Connected. Server Version = 8.0.30
21:12:35.867 -> INSERT INTO dev_poolcontrol.pc_log (proc, Status, Message) VALUES ('test sensor','xx','y')
21:12:35.867 -> [SQL] MySQL_Query::execute: Reallocate buffer, size =  95
21:12:35.867 -> [SQL] MySQL_Query::execute: query =  INSERT INTO dev_poolcontrol.pc_log (proc, Status, Message) VALUES ('test sensor','xx','y')**⸮+ǎ⸮⸮⸮⸮N=⸮⸮⸮⸮@⸮⸮⸮,ׅ⸮䤟⸮⸮׺⸮r}⸮⸮⸮FY⸮⸮Ȋ.**
N
21:12:35.867 -> [SQL] MySQL_Query::execute_query: query =  INSERT INTO dev_poolcontrol.pc_log (proc, Status, Message) VALUES ('test sensor','xx','y')⸮+ǎ⸮⸮⸮⸮N=⸮⸮⸮⸮@⸮⸮⸮,ׅ⸮䤟⸮⸮׺⸮r}⸮⸮⸮FY⸮⸮Ȋ.
N
21:12:36.197 -> [SQL] MySQL_Packet::wait_for_bytes: OK, Num bytes=  14
21:12:36.197 -> [SQL] MySQL_Packet::read_packet: packet_len=  10
21:12:36.197 -> [SQL] MySQL_Packet::read_packet: exit
21:12:36.197 -> [SQL] MySQL_Packet::get_packet_type: packet type=  0
21:12:36.197 -> [SQL] MySQL_Packet::get_packet_type: packet type= MYSQL_OK_PACKET
21:12:36.197 -> Complex Data Inserted.
21:12:36.197 -> [SQL] Disconnected
21:12:36.197 -> 
21:12:36.197 -> Sleeping...
21:12:36.197 -> ================================================

11th insert attempt fails

21:13:07.635 -> Connecting...
21:13:07.635 -> [SQL] Connecting to Server: 192.168.0.10 , Port =  3306
21:13:17.675 -> [SQL] connected = 0
21:13:17.675 -> [SQL] Can't connect. Retry # 1
21:13:27.677 -> [SQL] connected = 0
21:13:27.677 -> [SQL] Can't connect. Retry # 2
21:13:37.680 -> [SQL] connected = 0
21:13:37.680 -> [SQL] Can't connect. Retry # 3
21:13:47.659 -> [SQL] connected = 0
21:13:47.707 -> [SQL] Can't connect. Retry # 4
21:13:57.680 -> [SQL] connected = 0
21:13:57.680 -> [SQL] Can't connect. Retry # 5
21:14:07.691 -> [SQL] connected = 0
21:14:07.691 -> [SQL] Can't connect. Retry # 6
21:14:17.680 -> [SQL] connected = 0
21:14:17.680 -> [SQL] Can't connect. Retry # 7
21:14:27.727 -> [SQL] connected = 0
21:14:27.727 -> [SQL] Can't connect. Retry # 8
21:14:37.713 -> [SQL] connected = 0
21:14:37.713 -> [SQL] Can't connect. Retry # 9
21:14:47.735 -> [SQL] connected = 0
21:14:47.735 -> [SQL] Can't connect. Retry # 10
21:14:47.735 -> 
21:14:47.735 -> Connect failed. Trying again on next iteration.
21:14:47.735 -> 
21:14:47.735 -> Sleeping...
21:14:47.735 -> ================================================

khoih-prog, 2022-12-07:
Hi @gmstuff 

Please carefully test your system, especially **code**, etc. before posting nonsense like this.

I've tested and everything is ON, on `Nano_RP2040_Connect`, using same WiFiNINA code, just a little mod to show count

```
uint32_t count = 0;

void loop()
{
  MYSQL_DISPLAY("Connecting..."); 
  Serial.print("Count = "); Serial.println(++count);
  ...
}
```

### Terminal

```
00:07:07.051 -> Starting Complex_Insert_WiFiNINA on MBED NANO_RP2040_CONNECT
00:07:07.051 -> MySQL_MariaDB_Generic v1.7.2
00:07:07.813 -> Attempting to connect to SSID: HueNet
00:07:10.267 -> SSID: HueNet IP Address: 192.168.2.116
00:07:10.267 -> Signal strength (RSSI): -25 dBm
00:07:10.267 -> Connecting to SQL Server @ 192.168.2.112 , Port = 5698
00:07:10.267 -> User = invited-guest , PW = the-invited-guest , DB = test_arduino
00:07:10.267 -> Connecting...
00:07:10.267 -> Count = 1
00:07:11.394 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:07:11.693 -> Complex Data Inserted.
00:07:11.693 -> [SQL] Disconnected
00:07:11.693 -> 
00:07:11.693 -> Sleeping...
00:07:11.693 -> ================================================
00:07:21.713 -> Connecting...
00:07:21.713 -> Count = 2
00:07:22.807 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:07:23.106 -> Complex Data Inserted.
00:07:23.139 -> [SQL] Disconnected
00:07:23.139 -> 
00:07:23.139 -> Sleeping...
00:07:23.139 -> ================================================
00:07:33.125 -> Connecting...
00:07:33.125 -> Count = 3
00:07:34.253 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:07:34.551 -> Complex Data Inserted.
00:07:34.551 -> [SQL] Disconnected
00:07:34.551 -> 
00:07:34.551 -> Sleeping...
00:07:34.551 -> ================================================
00:07:44.537 -> Connecting...
00:07:44.537 -> Count = 4
00:07:45.366 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:07:45.665 -> Complex Data Inserted.
00:07:45.665 -> [SQL] Disconnected
00:07:45.665 -> 
00:07:45.665 -> Sleeping...
00:07:45.665 -> ================================================
00:07:55.651 -> Connecting...
00:07:55.651 -> Count = 5
00:07:56.779 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:07:57.077 -> Complex Data Inserted.
00:07:57.077 -> [SQL] Disconnected
00:07:57.077 -> 
00:07:57.077 -> Sleeping...
00:07:57.077 -> ================================================
00:08:07.096 -> Connecting...
00:08:07.096 -> Count = 6
00:08:08.191 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
[... 167 lines cut ...]
00:16:18.791 -> 
00:16:18.791 -> Sleeping...
00:16:18.791 -> ================================================
00:16:28.763 -> Connecting...
00:16:28.763 -> Count = 50
00:16:29.889 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:16:30.187 -> Complex Data Inserted.
00:16:30.187 -> [SQL] Disconnected
00:16:30.187 -> 
00:16:30.187 -> Sleeping...
00:16:30.187 -> ================================================
00:16:40.184 -> Connecting...
00:16:40.184 -> Count = 51
00:16:41.310 -> INSERT INTO test_arduino.hello_sensor (message, sensor_num, value) VALUES ('test sensor',24,50.1)
00:16:41.608 -> Complex Data Inserted.
00:16:41.608 -> [SQL] Disconnected
00:16:41.608 -> 
00:16:41.608 -> Sleeping...
00:16:41.608 -> ================================================

```

and still running.

The issue will be closed and won't be reopen until a bug is clearly proven.

closed 2022-12-07 by khoih-prog